srg74/WLED-wemos-shield #2: Do you sell finished WLED-wemos-shield?

dioem1985, 2020-05-24:
Hello, sorry I choose this way to write to you. Do you also sell assembled WLED-wemos shields? Would love to buy a few from you. Do you have an email address? Would also leave mine here if they sold me one or more WLED-wemos-shield. I come from Germany and have only started soldering since last year.
Sincerely yours
dioem1985

srg74, 2020-05-24:
Hi. This is open source project and you can ask questions. As of right now I don't assemble and sell shield. All my intend was to help WLED community with affordable DIY solution. Especially for the beginners. Nova days most of the projects using SMT components and for the beginners very difficult handle them and expensive.
You can order PCB and parts very cheap. Talk to your friends DIYers and assemble few PCBs for yourself and friends - have fun. People doing classes for the assembly and fun. Old people who never hold soldering iron in their hands. Participate https://wled.discourse.group/ and you find people from Germany who already doing shield or think about it.
If nothing above is good for you. You can buy similar board from Quinled.
Best regards.

dioem1985, 2020-05-24:
Thank you for your reply

dioem1985, 2020-05-25:
Hi,
I just ordered the pcb from pcbway, thanks for that. Can you please give me a precise list of the parts you need, with the exact description of the parts? Thank you very much for your effort.
Sincerely yours
dioem1985

srg74, 2020-05-25:
Did you navigate repository? https://github.com/srg74/WLED-wemos-shield/blob/master/resources/BOM.pdf

dioem1985, 2020-05-25:
Unfortunately I missed that.  Thank you very much.  Sorry if I annoy you so much.

srg74, 2020-05-25:
Np

srg74, 2020-06-14:
Solved

closed 2020-06-14 by srg74

Joshfindit, 2022-03-07:
I found this issue by searching for PCBWay, and wonder if you could share the complete project through https://www.pcbway.com/project/shareproject/

Two benefits:
1. People would be able to order PCB and all parts at the same time
2. There could be an SMD version that people could order assembled for extra cost

netmindz, 2022-03-13:
+1 for shared project