srg74/WLED-wemos-shield #4: 1N4148 Digikey does not exist

WoodyLetsCode, 2020-06-12:
Hello,
in your BOM the Digikey of 1N4148 does not exist.
Can you update this?
Also it would be nice if you create a shopping cart link for digikey. :)

srg74, 2020-06-12:
Just for now part# is 1N4148-P-TAPGICT-ND. Later I will update BOM

WoodyLetsCode, 2020-06-12:
I ordered [this](https://www.digikey.de/product-detail/de/1N4148/1N4148FS-ND/458603?utm_medium=email&utm_source=oce&utm_campaign=3026_OCE20RT&utm_content=productdetail_DE&utm_cid=1343331&so=64601622&mkt_tok=eyJpIjoiTXpoaU9Ea3dZMkppTVdaaSIsInQiOiIrbmxNXC9MVkltYjZkZTVucTI4XC9LdWJ6UFdpc3FWelpsS3NDem1wYmlaaVcxWm1jYmx6bGJCMVpsRXlzdnloXC9tbVpVSkpjSDhSZWxkakhoWlFpMHNNbFl4cDhEdVlzMit5dndsa0NTWW1rXC9cL0NIcEY1SDBjaEFvRkZ6OGxCVnNNIn0%3D). Can I also use it?

srg74, 2020-06-12:
BOM corrected

closed 2020-06-12 by srg74