srg74/WLED-wemos-shield #5: 10k OHM Resistor too big

WoodyLetsCode, 2020-06-22:
Hey,
it seems that the 10k OHM Resistor is a little bit too big for the holes.
[image]
It's quite a struggle to get it fit into the holes.
[image]
The 4.7k OHM Resistors have the perfect size for the holes.
Maybe you choose 10K OHM resistors in the size of the 4.7k OHM resistors in your BOM, if possible. 😄

srg74, 2020-06-26:
Thank you for your report.
BOM now is updated to correct values https://github.com/srg74/WLED-wemos-shield/blob/master/resources/BOM.pdf

WoodyLetsCode, 2020-06-26:
Okay, thank you! 😄

closed 2020-06-26 by WoodyLetsCode