srg74/WLED-wemos-shield #6: Wrong value for F2 & other PTC fuse?

WoodyLetsCode, 2020-06-28:
Hello,
the value of F2 (RUSBF075-APHFCT-ND) in your BOM is _30V 500mA_, but it is a **6V 500mA** fuse.
And aren't 6v a little bit too low? For example when somebody accidentally connects it to 12V :sweat_smile: it is possible that electricity arcs over the fuse.
Wouldn't this PTC fuse be better?
DigiKey: [507-2375-ND](https://www.digikey.de/product-detail/de/bel-fuse-inc/0ZRG0075FF1E/507-2375-ND/9468224)

srg74, 2020-06-28:
Thank you for your correction. Fixed in update.

WoodyLetsCode, 2020-06-28:
How do you make this BOM pdf file?

srg74, 2020-06-28:
I'm a Mac user so using a Numbers for spreadsheets. On DigiKey website I'm creating BOM file then exporting as CSV and then editing in Numbers to what you see. After just instead of printing save as PDF. Done!

WoodyLetsCode, 2020-06-28:
Can you also export it as .XLSX-File, so other users can edit it with excel?

srg74, 2020-06-28:
I don't who else need to edit this file, but I did a copy in XLSX

WoodyLetsCode, 2020-06-28:
Thank you! :smile:

closed 2020-06-28 by WoodyLetsCode