srg74/WLED-wemos-shield #10: Overvoltage protection

WoodyLetsCode, 2020-07-05:
Hey,
are you interested in adding an overvoltage protection?

srg74, 2020-07-05:
Over voltage protection is not simple circuit and no TH version. First it will make project more expensive and more complicated. Both issues is against idea to make as simple as possible and not expensive.

WoodyLetsCode, 2020-07-06:
What if you use this part as DC DC converter:
Digikey: [1589-1488-ND](https://www.digikey.com/products/en?keywords=1589-1488-ND)

This would have some advantages:

1. We don't need a power selector any more
2. It doesn't matter if you connect 5V or 12V power supply
3. You cannot destroy your WEMOS by setting power selector to wrong voltage

srg74, 2020-07-06:
> What if you use this part as DC DC converter:
> Digikey: [1589-1488-ND](https://www.digikey.com/products/en?keywords=1589-1488-ND)
>
> This would have some advantages:
>
> 1. We don't need a power selector [...]

Looks like you found a golden drop in replacement! But only one downside is price :( So actually we can add as more expensive alternative and eliminate header for selector and add a option to make a permanent wire jumper for universal voltage input. If you want you can add this option to Wiki and I will change BOM part as alternative. Only my worry is very low current 300mA. For ESP32 maybe near a limit, they very hunger sometime :)

WoodyLetsCode, 2020-07-11:
> 
> 
> Over voltage protection is not simple circuit and no TH version. First it will make project more expensive and more complicated. Both issues is against idea to make as simple as possible and not expensive.

Hey, over voltage is not very complicated. You can use a [Gas Discharge Tube Arrester](https://www.digikey.com/products/en/circuit-protection/gas-discharge-tube-arresters-gdt/142?k=&pkeyword=&sv=0&pv69=411897&sf=1&FV=-8%7C142&quantity=&ColumnSort=0&page=1&pageSize=25) or a [Varistor](https://www.digikey.com/products/en/circuit-protection/tvs-varistors-movs/141?k=&pkeyword=&sv=0&pv69=411897&sf=1&FV=-8%7C141&quantity=&ColumnSort=0&page=1&pageSize=25) or both. 😄 Both parts are not very expensive. They are not very complicated to solder. You can mount them through hole. And now we have options so you can make it optional to add an overvoltage protection.
There are many easy guides in the internet how to use a Gas Discharge Tube Arrester or Varistor.
I hope this is useful. 😃

srg74, 2020-07-11:
I would agree with you if proposed parts will be used to protect hi voltage AC input 120-240V :)

WoodyLetsCode, 2020-07-11:
Yes with this parts you can protect 120-240V.

WoodyLetsCode, 2020-07-11:
With  Gas Discharge Tube Arrester you can only protect from DC Overvoltage. But with a Varistor you can protect from both.

closed 2020-09-02 by srg74