srg74/WLED-wemos-shield #9: Change text for fuse holder

WoodyLetsCode, 2020-07-05:
Currently the text on the PCB where the fuse holder is placed is "FUSE".
Can you change this to "F1"? I think this would be better. :smile: 
There's no hurry, but if you want you can change this when you plan your next release.

WoodyLetsCode, 2020-07-06:
I think this is better, because on any other part you did wrote the short name. And this is good because so you can easily look into the BOM and see what part you have to solder. So I don't understand why you only wrote "FUSE" for instead of "F1". I know it's looks logical that you have to solder F1 because of the shape, but I think for people who solder the first time parts on a PCB it could be a little bit confusing. And you yourself want to make this project as easy as possible.

srg74, 2020-07-06:
Will be fixed in next update.

WoodyLetsCode, 2020-07-08:
Thanks :smile:

srg74, 2020-07-11:
Updated fuse holder text.

WoodyLetsCode, 2020-07-11:
Can you publish the new release? 😄

srg74, 2020-07-11:
Done

closed 2020-07-11 by WoodyLetsCode