srg74/WLED-wemos-shield #14: esp32 - seems not to work with wled_base_2007020.bin

Jojopirat, 2020-09-02:
I flashed my ESP32 (form factor Wemos D1 Mini) with wled_base_2007020.bin from https://github.com/srg74/WLED-wemos-shield/tree/master/resources/Firmware/0.10_2007020/ESP32. Wifi and the web interface work, but neither the relay nor the OLED work. With an ESP8266 at least the relay works. I haven't been able to test LEDs yet.
What do I have to consider if I want to use an ESP32 instead of an ESP8266 and which OLED display is supported?

srg74, 2020-09-02:
I'm sorry for your inconvenience. Looks like file wled_base_2007020.bin is corrupted. Please use tested wled_base_2008310.bin
OLED display is no name 0.91" I2C similar to this https://github.com/srg74/Controller-for-WLED-firmware/blob/master/resources/controller.jpg.
ESP32 working just fine with amazing WLED firmware. Matter of fact it's a personal preference what people use.

Jojopirat, 2020-09-02:
Thank You for support.
With wled_base_2008310.bin on ESP32 relay and OLED-Display work fine now.

srg74, 2020-09-02:
I just posted fresh bin file with minor update. If you wish you can try this one. And you can connect button too :)
I assume you using my shield design.

Jojopirat, 2020-09-03:
I have now tried wled_base_2008310.bin with Led strips. It doesn't work with the ESP32. With the ESP8266 it works with clockless strips. Stripes with clock do not work on both CPUs.
Do you have an idea why?
I have than installed 0.10_2005030 on the ESP32. After this clockless led stripes work. But ESP32/8266 don't work together with my strip with the clock line.
Last xmas I used this four wire stripe without your board and with self compiled wled firmware on an 8266 wemos d1 mini. So i had to configure pinout and typ of stripe in the source code. For this reason i wonder about firmware on your repository doesn't make a difference.
May I have to compile wled on my own to support my configuration?

srg74, 2020-09-03:
Bin files located in repository support only 3 wire addressable strips. Those is only a favor for people who is using board. I don’t sell them. So my board as also WLED project is open source and all changes in configurations should be addressed by users due to endless variants of the setup.
You definitely can configure your bin file accordingly. In this case look at schematic and you will find pinout table for shield.
Best regards.

Jojopirat, 2020-09-03:
I got it wrong. The information about the hardware supported by the board also included the 4 wire stripes. I wrongly assumed or had hoped that this also applies to the firmware. Your project is very good and I am delighted that you are sharing it with us. I also didn't expect that you would provide firmware for every variant, but I did not want to reinvent the wheel if there is already a variant of the firmware that fits my configuration.
So thank you again for sharing and your support.

srg74, 2020-09-03:
Thanks for a good words! Indeed board support variety of configurations including 4 wires LED strips. Like I said user can choose what needed to be connected I just provided possibilities. Plus I don’t have that many different LEDs to test every possible configuration. I can only provide bin files what I can test. Also a lot of people already using board and I know most of them using WLED discord channel and Discourse so if anyone have questions they can ask there too.
Again thank you for choosing my design :)

closed 2020-09-09 by srg74