srg74/WLED-wemos-shield #15: WROOM32 not wokring, wrong data pin

neilpryde1984, 2020-09-03:
Hi

Just tried out today my WLED Wemos shiel, but got only one, white LED lighting. Connecting to the build in AP was successful, but changing led type and/or other settings didn't help.

After some troubleshooting, I figured out, that for this board, the datapin in the firmware is wrong. when i connect the leds datapin to D4 of the board (https://bit.ly/3gWNehp), the whole thing is working as expected.

I do not think, that this is a hw issue, I've 4 identical boards, all with the same behaviour..

Regards
Chris

srg74, 2020-09-03:
What version of the board are you using? Then check out schematic first. Data pins for Wemos D1 mini and ESP32 mini version is not same GPIOs. https://github.com/srg74/WLED-wemos-shield/blob/master/resources/schematic.pdf In schematic you can find pinout for both main board which can be plugged to shield.
You have to use custom bin file for ESP32 board to use with shield https://github.com/srg74/WLED-wemos-shield/tree/master/resources/Firmware/0.10_2008310/ESP32

neilpryde1984, 2020-09-03:
The board has a label "MIN132 V1.0.0" on it. according to the pin description on the board an also its shematic (https://ae01.alicdn.com/kf/HTB1mlmbdGWs3KVjSZFxq6yWUXXa2.jpg),  the D4 from the shield is the IO16. This corresponds with your shematics and agrees for the ESP32 firmware..

I'm sorry about such a "noob-like" question.. but i would'n ask, if I had not multiple boards with the same issue..

srg74, 2020-09-03:
There is no noob question :) You asking and it's good. Please let me know what version of shield are you using, also where did you download firmware file e.g. bin file.

neilpryde1984, 2020-09-03:
I'm using shields build up according your guide for the Wemos wled shield (
https://github.com/srg74/WLED-wemos-shield)
The PCBs were ordered a few days ago..

The current installed firmware is
https://github.com/srg74/WLED-wemos-shield/blob/master/resources/Firmware/0.10_2008310/ESP32/wled_base_2008310.bin

Except of the data pin, all is working fine (relais, display,..)
Unfortunately, I do not have any oscilloscope for checking what signal the
data pin on the header an screwblock has..

srg74, 2020-09-03:
Could you please re-download bin file and let me know if it working.

neilpryde1984, 2020-09-04:
thanks for the new firwmware. but it seems to be a step backwards.. now, display and ap are also "failing"/not working

srg74, 2020-09-04:
Just checked again bin file with same kind of board as you have and everything is working file.
[image] Please erase your board and re-check all connections.

neilpryde1984, 2020-09-05:
A reset a day, keeps the troubles away ;)

The erase helped!! Now, everything is worjing fine with the latest ESP32
bin.

Thanks for your help!

closed 2020-09-09 by srg74