srg74/WLED-wemos-shield #16: What is the maximum load i can connect to this?

Bosken85, 2020-09-07:
Hello,

First off, nice work! Really appriciate the time you've taken to make this. I was wondering what the maximum load is i can connect to this board. I have long SK6218 led strip that consumes a max of 15A. But i noticed the Relay has a max of 10A. Do i need to replace some parts of the board with higher capacities?

Kind regards
Kevin

srg74, 2020-09-07:
You absolutely correct about relay ratings. I don’t think you will find relay with similar footprint and higher current rates to replace on the board. Instead what you can do it’s connect external relay with higher ratings and switch with it. Also if you count Amps for your strip 60mA per LED this case almost never happened. So you can run without any modifications just for safety limit in WLED current max to 10000mA

WebOscar, 2020-09-30:
I found a 20A Relay, which should fit to your needs: https://www.aliexpress.com/item/32842347863.html?spm=a2g0o.detail.1000023.15.21953b72Bjcqy7
But I would also order the Boards with a thicker layer of copper. At least I did with 2 oz Cu to handle higher ratings and get less loss.

Greetings,
Oskar

srg74, 2020-09-30:
Very good find! I don’t know if this relay will hold actually 20A but there’s a possibility :) please be very cautious when using big load.

WoodyLetsCode, 2020-12-12:
Hey,
i added a FAQ for this:
https://github.com/srg74/WLED-wemos-shield/wiki/FAQ#what-is-the-maximum-load-i-can-connect-to-this

shortbloke, 2021-01-09:
@srg74 Hi I've just taken delivery of some of the 20A relays suggested and their pin spacing is 10mm rather than 12.2 used with the original 10A relay. So not a drop-in replacement with your current PCB layout. Probably can be made to fit with some carefully drilling of new holes. But a bit of a bodge.

srg74, 2021-01-09:
> @srg74 Hi I've just taken delivery of some of the 20A relays suggested and their pin spacing is 10mm rather than 12.2 used with the original 10A relay. So not a drop-in replacement with your current [...]

Sorry for wrong information about relay. Unfortunately I don't tested myself. On of project user recommend as drop in replacement. Can you post exact dimensions or data sheet and I will see what can be done for future board.

shortbloke, 2021-01-10:
[image]

[image]

This comes from the AliExpress link. Though section 4 mentions both 20A and 10A as the rating, I'm not sure I'd trust this at 20A, it seems so small!

srg74, 2021-01-11:
I have read data sheet and I can say it's not a 20A relay but only 15A.

MFornander, 2021-01-26:
> I found a 20A Relay, which should fit to your needs: https://www.aliexpress.com/item/32842347863.html?spm=a2g0o.detail.1000023.15.21953b72Bjcqy7

Just another note that I went ahead and ordered the relays here and the spacing is indeed too small, just in case it was just a bad shipment.  So link above should not be used.

closed 2021-04-01 by srg74