srg74/WLED-wemos-shield #18: Question about Fuse F1

WebOscar, 2020-09-30:
At first, thank you so much for your great work! I just ordered 150 boards for a lot of own Smart-Home-projects and for some Friends, who are such nerds! ;) And 10 sound reactive boards for testing. I hope PCBWay will pay you the promised 10% as I have not found your "Buy me a Beer"-Button! ;)

I have a question about fuse F1. Unfortunately the BOM does not list which fuse should be inserted there. Don't I have to calculate down the ampere of the 32V fuse for a voltage of 5V?

If I calculate with a maximum of 10A at 5V, because the relay should not switch more, then I think that I have to convert the corresponding values of the fuses, which are indicated with a nominal voltage of 32V, or am I wrong?

With 10A, 5V I get 50W which is about 1,6A at 32V. The next suitable fuse would be 0297002.WXNV (Digikey), right?

Greetings from Germany,
Oskar

Translated with www.DeepL.com/Translator (free version)

srg74, 2020-09-30:
Actually fuse rating have to be calculated like this Calculated maximum Amps X 1.25 = Fuse Amps. 32V DC is maximum that fuse can hold. Those fuses is automotive so it nom 12VDC. 
You are very kind to your friends :) Thank you for your generous words

WebOscar, 2020-09-30:
Oh dear, now I'm confused!

Okay, let's say 10A, 5V is the max I want to protect. 10A x 1,25 = 12,5A * 5V = 62,5W

What are the 12,5A now for the automotive fuses?

Lets assume that the specifications on the fuse refer to a nominal voltage of 12V. Will I have to use 62,5W / 12V = 5,2A for using this fuse with 5V (round up or down to the next suitable value of a fuse) or will I use a 15A fuse (I think, there's no other between 10A and 15A) as calculated for 5V, regardless of the nominal voltage of the fuse?

Do you know, what I mean?

WebOscar, 2020-09-30:
Okay! I found the answer! The fuse is limiting the current, regardless of the voltage. The 32V in the example above only gives the safe "isolated up to 32V" value.
Learned a lot about fuses, today! Thanks for your fast answer!

srg74, 2020-09-30:
Let me clarify. With fuse you protecting your circuit from over current. So in general it’s related to your wire gauge (AWG). Now after you calculate proper wire gauge for your project it’s tell you what Current it will can carry. Now you can multiply that number to 1.25 and result will tell you what nominal Amps fuse have to be.

WebOscar, 2020-09-30:
Oh yes, that's clear; thanks! I only got lost in calculation as the fuse came up, which wasn't necessary at all, as I know now.
One last Question (I hope). Is the PS1 - as I understand is a step down converter - needed when it is clear, there will be only 5V used?

srg74, 2020-09-30:
Yes step down buck converter is optional. You can build only a with components is necessary for your project functionality. Starting from https://github.com/srg74/WLED-wemos-shield/wiki/Bare-minimum and build up to whatever fit your project.

WebOscar, 2020-09-30:
Oskar, RTFM! Dammit! Thanks for the link. I missed that! Now I have a lot of capacitators and resistors left!
RTFM, RTFM, RTFM, RTFM...

closed 2020-09-30 by WebOscar