srg74/WLED-wemos-shield #20: Audio jack replacement

rboers72, 2020-09-30:
I want to build the audio version of the WLED-wemos-shield and I have some PJ-210B audio connectors lying around. Is there any reason why I shouldn't use these instead of the SJ1-2503A?
I compared them, and they look the same.
The PJ-210B is a perfect fit in the PCB.

[image]

huggy-d1, 2020-09-30:
At a glance, the physical dimensions of the pins are slightly different, but with hole tolerances, I can see why you say it's a perfect fit.  The soldering of the device may be a tiny bit trickier to get proper flow through the holes to achieve expected mechanical support.  That said, if this is a connect it and forget it install, that's probably not a problem.

If you expect unplugging the connector a lot, I recommend using a sacrificial cable/connector.  Plug in the very short sacrificial cable into the PJ-210B and leave it connected all the time, and simply use the other end of the cable as the "field" side of the connector.  This also applies to many connections you expect to insert / remove often.  Using this approach protects the life of PCBs and original connectors, and instead wearing out an easily replaceable cable.

rboers72, 2020-09-30:
@huggy-d1 Thank you  for your reponse. I'm going to use this one, because I'm not going to connect/disconnect it a lot.

closed 2020-09-30 by rboers72