srg74/WLED-wemos-shield #21: Sound reactive board - bare minimum

WebOscar, 2020-10-01:
Hello again!

I couldn't find any extra bare-minimum documentation for the sound reactive board. (Or does it exist and I am just too damn dumb to see it?) So I compared the two BOMs to identify the extra parts the SR-board needs.

This is my result:

R8, R9, R10, R11, R12, C3, C7, F2 (different than the standard version), Audio Socket and SW1.

Am I right or did I miss something?

Greetings from Cologne, Germany
Oskar

srg74, 2020-10-01:
There is no guide for SR board,  since difference is only circuit for microphone and linear input. You found differences correct, just F2 is part of base circuit - it’s a fuse

WebOscar, 2020-10-01:
Thanks! I know F2 is a fuse, but according to the BOMs it is a different one as in the standard board. Standard board BOM lists a 16V 900mA while sound reactive BOM lists a 30V 500mA. That's why I marked it in my findings as different from standard, even I don't think it will make a difference.

closed 2020-10-01 by WebOscar