srg74/WLED-wemos-shield #25: v1.3 Not working. What I have done wrong?

waterfox207, 2020-11-21:
Hello,
and thanks for the nice project. 
I have built the board in Version 1.3. 
Sadly, it is not working, and I have no voltage output on the output pins.

So far, I have checked PIN D6 which should be the relay pin. If the wemos is not connected to board, I get a high and low signal (0 vs. 3.3 V). If I put the wemos on the board, I get 0 V and 0.4 V. 
If I disconnect the board from the power supply and check for continuity D6 is connected to ground. I think this is not correct? 

I have also added the switch to the board. All pins of the switch also have continuity with ground. 
Furthermore, I do not know if JP3 needs to be connected or not? 

What is the purpose of D3<BT>?

I hope someone can help me 
[image]

WoodyLetsCode, 2020-11-21:
Hello,
did you check voltage between 5V Pin and GND Pin?
D3 is for a push button.

waterfox207, 2020-11-21:
Thank you for your fast response!
I about 5V between 5V Pin and GND Pin.

I thought SW1 is for the push button? Thats the place where my button fits.

WoodyLetsCode, 2020-11-21:
Did you turn on the LEDs in WLED?

waterfox207, 2020-11-21:
Yes, the blue led on the wemos also indicates the status of the led output. If the small blue led is on I think the lights are also on?

WoodyLetsCode, 2020-11-21:
No, the blue LED on the wemos is always on. You have to turn on the LEDs in the Web-GUI. Do you hear a "click" from the relay noise when you connect it to power?

waterfox207, 2020-11-21:
When I connect the board to power. There is no noise from the relay. 
Hmm I also think there is some "magic" going on. When I now place the wemos on the board the blue light only flashes once. 
I have also no access the the Access point of the wemos.
If i connect the wemos via usb, everything works fine. 
Maybe I have done a pull down resistor wrong?

WoodyLetsCode, 2020-11-21:
Can you send a picture how you put in the wemos?

waterfox207, 2020-11-21:
[image]

srg74, 2020-11-21:
1. Upload code to you Wemos board and make sure everything is woking fine without shield. Means connection to your network and navigation by GUI.
2. Check out your board without Wemos connected. Make sure board have no shorts. It have to be 5V on headers marked 5V (I assume your board is 5V only).
3. Plug your Wemos. If everything done correct (you check your boards and firmware is correct) than upon power ON you can hear a click from relay, means it's working correct. Now shutdown power and connect your LED strip to board. Power ON and yours LEDs have to be orange.
Basically board don't have a lot of parts and it's very simple to repeat (proven by many users before)

waterfox207, 2020-11-21:
I found my mistake!
For R3 I have used a 10 Ohm resistor and not a 10k Ohm one. 🙈
I will tell you my result when I have changed it. 

I also have now removed the button. Because of the continuity check. So far, the wemos is now also working on the board and D3 isn't shorted to ground anymore. I think maybe there is a misstake in the gerber or my button does not work like the ones from digikey. My button is connected to ground on the housing. So, both bigger pins are connected. I think that this is the reason for the short circuit on the d3 pin.
[image]

Thank you for the hint with the fuse. I was not able to order the fuse holder from the BOM. When I receive the right fuse holder, I will replace it.

srg74, 2020-11-21:
Definitely SW1 have a gerber error. I will fix that immediately. Thank you.

waterfox207, 2020-11-21:
Perfect. 
Thank you for this nice Projekt. 🤙

closed 2020-11-21 by waterfox207