srg74/WLED-wemos-shield #26: Radial Fuse Alternative

Ryanb369, 2020-11-22:
First off, thanks for making this awesome solution and allowing it to be open! I've got a question about the 16v radial fuse. Is there an alternative to this, I'm finding the availability of these are slim. Thanks in advance!

srg74, 2020-11-22:
Why they not available?

Ryanb369, 2020-11-22:
I may have been looking in the wrong places, I appreciate the quick response. Would a 16v 1A Work?

srg74, 2020-11-22:
I guess no harm. Main concern is time it should trip. Shorter time is better.

Ryanb369, 2020-11-22:
Gotcha, so within 500-700mA is best practice?

srg74, 2020-11-22:
Yep

zsolesz79, 2020-11-27:
I'm struggling to find some of the components locally...  :( One of them is the PTC, so how much would be the suitable time to trip value that I should look for?

Another one is the 10uF ceramic capacitor? Can I go with a tantalum or a low ESR electrolytic instead?

closed 2020-12-06 by srg74