srg74/WLED-wemos-shield #27: Bare minimum - Not working

Dragi2k, 2020-11-24:
Hello,

i have Build the Board and its not Working.
Is it Possible that "R8" is needet ?

WoodyLetsCode, 2020-11-24:
Hello,
R8 is not needed for Bare minimum. Can you give more informations? What exactly is not working? Can you send a picture of your assembled board? What are the results of the [check](https://github.com/srg74/WLED-wemos-shield/wiki/Check) steps?

Dragi2k, 2020-11-24:
Nur if i shorten r8 it works fine 🤔
[image]

Dragi2k, 2020-11-24:
[image]
[image]
[image]
And the Fuse holes are also wrong, but i dont know, ob this false by pcb-way is

srg74, 2020-11-24:
Some how resistor R8 150 Ohm. You do need this resistor for minimum configuration. Look at BOM v1.3 because your board is v1.3. You can find archive here https://github.com/srg74/WLED-wemos-shield/releases/tag/v1.3

Dragi2k, 2020-11-24:
I know, fuse is in the way 😉
But what is with the holes of the Fuse holder?
[image]

srg74, 2020-11-24:
Where did you order your PCB?

srg74, 2020-11-24:
[image]

Dragi2k, 2020-11-25:
Over your link, on pcb-way

srg74, 2020-11-25:
In this case you have contact PCBway and explain that your board came not routed. They have repeat your order it's their responsibility.

closed 2020-11-25 by Dragi2k