srg74/WLED-wemos-shield #28: Resistors R1 / R8

Ryanb369, 2020-11-24:
i may be blind, but the value for R8 is not listed on BOM. I can't seem to find R1 on the board either. Thanks for the info in advance.

WoodyLetsCode, 2020-11-24:
Hello,
R1 is next to the 8-Pin header. Where do you see R8? I think R8 doesn't exist, or am i wrong?

Ryanb369, 2020-11-24:
[image] R8 is right in the middle, could that be R1?

WoodyLetsCode, 2020-11-24:
Oh sorry. I don't have the new v1.5 board. I just have the older ones. But I'm sure @srg74 can help here. 😃

srg74, 2020-11-24:
Resistor you asking is not R8 but R2 150 Ohm

Ryanb369, 2020-11-24:
Is there a value for R8? Or is this not needed

srg74, 2020-11-24:
Please read right above :)

srg74, 2020-12-06:
Errors fixed

closed 2020-12-06 by srg74