srg74/WLED-wemos-shield #29: D1 not booting when connected to 12v

Ryanb369, 2020-11-25:
The D1 blinks once and goes dark. Led indicator on wemos-shield  is lit up. D1 is fine when connected via usb. Unsure of why, [image]

[image]

WoodyLetsCode, 2020-11-25:
Hello,
what are the results of the [check](https://github.com/srg74/WLED-wemos-shield/wiki/Check) steps?

Ryanb369, 2020-11-25:
All the voltages are as the should, at jumpers and board inputs.

srg74, 2020-11-25:
Please remove push button from the board. Early batch of PCBs have error. If you comfortable to cut traces I can explain what to cut.

Ryanb369, 2020-11-25:
I'll desolder the button, i believe i have pcb 1.5

srg74, 2020-11-25:
So is it working now?

srg74, 2020-11-25:
Also for quick response you can reach me https://discord.gg/KuqP7NE  as @Serg74

closed 2020-12-19 by srg74

kallejoken, 2024-08-06:
> So is it working now?

I have exactly the same problem with version 3.0, I didn't solder the push button. Do you have any help?

srg74, 2024-08-06:
Version 3.0 don’t have errors and works for 2 years without problems

kallejoken, 2024-08-06:
> Version 3.0 don’t have errors and works for 2 years without problems

yes exactly, but only if I apply the fix as described here https://github.com/srg74/WLED-wemos-shield/issues/107

srg74, 2024-08-07:
Problem here is most likely due to different design of mini board used. But in this post you can find solution for the issue.