srg74/WLED-wemos-shield #30: R2 required for v1.5 bare minimum

SvenFestersen, 2020-12-06:
First of all thanks for this awesome shield! I just assembled my first v1.5 bare minimum board + temperature sensor. However, I was not able to turn the LEDs on on the first try. A look at the schematic revealed that R2 is actually required as it connects the level shifter output to the screw terminal. I soldered a 120Ohm resistor in that place (no 150 Ohms at hand) and it now works like a charm.

So I suggest updating the "Bare minimum" page to include R2.

srg74, 2020-12-06:
New WiKi in the work as of right now. So your remark will be reflected there.

WoodyLetsCode, 2020-12-12:
Hey,
I updaded wiki. Now R2 is mentioned in [bare minimum](https://github.com/srg74/WLED-wemos-shield/wiki/Bare-minimum).
I also added a FAQ for this:
https://github.com/srg74/WLED-wemos-shield/wiki/FAQ#do-i-need-r2

closed 2020-12-19 by srg74