srg74/WLED-wemos-shield #31: Are you able to update the shield box v1.5 to thingaverse?

o1hitman1o, 2020-12-11:
I can only locate the WLED wemos shield box v1.0 and not the latest one with the png you uploaded 12/9/2020

srg74, 2020-12-11:
Not yet updated. New version in the work.

srg74, 2020-12-15:
Here is link to new box https://www.thingiverse.com/thing:4686098

ubergeekseven, 2020-12-15:
Better yet, I see the board rendered bare in the files list. Do you happen to have the board populated and if so, could you export a step version of that? 
I received my 15 boards in less than A week and all my components arrived. If you had a model of the fully populated board available, I could create several revisions easily. I first made several boards by hand and created a nice car for them but, your pcb is very well laid out and compact. I want to create several designs with the options available as the differences.
Thanks for the work you've put in already. Love it all so far.

closed 2020-12-19 by srg74