srg74/WLED-wemos-shield #33: Update the BOM

Dragi2k, 2020-12-21:
Hallo,

is it Possible to Update the BOM, to show wich Part you need for wich Option ?

Because you Updatet your list but if you have an older version it dosent match anymore.

I have done it for the 1.3 version 
[BOM 1.3 Update.xlsx](https://github.com/srg74/WLED-wemos-shield/files/5724327/BOM.1.3.Update.xlsx)

srg74, 2020-12-21:
Thank you for your proposal. All options is reflected in WiKi. BOM is designed to upload to Digikey website and order only parts you need for your build. with additional fields in BOM Digikey website will not be showing correct order.

Dragi2k, 2020-12-21:
ok, but the wiki is also incorrect
on 1.3 :
R1 = not on the board
R2 = is there for R1
R8 = is there for R2 

on the other boards it may be different.

means on the wiki should be a change log, to see what has changed on the parts

Dragi2k, 2020-12-21:
and the bare minimum site is not linkt direkt in the guide
you must go over - order parts to enter the site

srg74, 2020-12-21:
> ok, but the wiki is also incorrect
> on 1.3 :
> R1 = not on the board
> R2 = is there for R1
> R8 = is there for R2
> 
> on the other boards it may be different.
> 
> means on the wiki should be a change log, to see what has changed on the parts

To find correct BOM you have download release v1.3 from here https://github.com/srg74/WLED-wemos-shield/archive/v1.3.zip

Dragi2k, 2020-12-21:
Yes i know, i have the file.
but on the file is no list wich parts you need for what. and that changes on your updates.

so on i am vary happy with the board, i appreciate that

srg74, 2020-12-21:
> Yes i know, i have the file.
> but on the file is no list wich parts you need for what. and that changes on your updates.
> 
> so on i am vary happy with the board, i appreciate that

Your file is added to repository https://github.com/srg74/WLED-wemos-shield/blob/master/resources/BOM.1.3.Update.xlsx

closed 2020-12-21 by Dragi2k