srg74/WLED-wemos-shield #36: Where is Dallas Temperature?

djashjones, 2021-01-15:
I've some boards with all the options and was under the impression that the temperature would be displayed in the info section as well as mqtt (not tested).

I've uncommented the 

  #For Dallas sensor uncomment following 2 lines
    OneWire@~2.3.5
    milesburton/DallasTemperature@^3.9.0 
    
    Complied for both d1 mini (clone) and my ESP32 and can't see it anywhere

srg74, 2021-01-15:
Uncommenting libraries is not enable usermod. Better if you use precompiled bin files from this repository. If you want to compile your code use my WLED fork this branch https://github.com/srg74/WLED/tree/WLED_wemos_shield

djashjones, 2021-01-15:
I've been using your fork as I could not see it using your binary. I can see the temperature via mqtt, so I know it's working.
[image]

srg74, 2021-01-15:
Yes it’s work only over mqtt. I don’t implement yet GUI part.

djashjones, 2021-01-15:
Awesome. At least it's not something I did. Love your shields. Got some outdoor ones to build for the summer.

closed 2021-01-19 by djashjones

yousaf465, 2022-04-19:
@srg74 how do I setup mqtt for this? which topics to use?