srg74/WLED-wemos-shield #38: Shelly RGBW2 OTA and button not working

aleksandrov1995, 2021-02-06:
Hello, I have successfully flashed the shelly wled firmware. However, I cannot get the button to work or the OTA update. The OTA fails with tasmota and wled firmware. Please advice.

WoodyLetsCode, 2021-02-06:
What shield version do you use?

aleksandrov1995, 2021-02-06:
Hi, I was unaware that I needed the shield, I am using the rgbw2 standalone. Although, this should not affect OTA.

srg74, 2021-02-06:
Hello. Definitely you don’t need shield. As I understand you found firmware for Shelly device. This is curtesy firmware for specific device. Your Shelly device might have different pinout. Regarding OTA - WLED is not supporting OTA for devices with memory less than 4mB.

aleksandrov1995, 2021-02-06:
Hi, thanks for the input. I am using your  firmware from this repo. So, the build in inputs for the shelly are not mapped into it? Noted about the OTA.

srg74, 2021-02-06:
It's mapped but as I mentioned your shelly might have different pinout. You can try compile yourself from here https://github.com/srg74/WLED/blob/ShellyRGBW2 This branch is a source for a bin file located here.

aleksandrov1995, 2021-02-06:
Thanks, that's all the info I needed.

closed 2021-02-06 by aleksandrov1995