srg74/WLED-wemos-shield #39: PCB v2.0 setup issue

ChristophA, 2021-03-28:
Hi,
thank you very much for this cool project! I recently ordered come PSBs v2.0 and the required parts. While assembling a minimum setup I struggled to find the correct place for the fuse F2. There's no lable on the PCB for F2. But after a while i figured out to place in on the right hand side of D2. Maybe it's worth to add a lable for F2 on the PCB!?

A Wemos D1 Mini (also tested with D1 Mini pro) is now starting and accessible viw the webinterface. I confiugre the relay to use pin / GPIO 12, works as expected. LED out (DI) is configured to use pin / GPIO 2. But I cannot light any LED. Not sure if my setup is correct. I've also added R1 and R2, but still no light.

Is there anything missing?

Many thanks for any hint!

PS: Still waiting for the fuse holder, therefore I'vs added a short cable just for testing.

[image]

srg74, 2021-03-28:
First  of all your IC is different from which project is using. 74HCT126 is a buffer not a shifter. I don't think it will work.

ChristophA, 2021-03-28:
I've double checked everything 10 times. Everyting but this chip...
Thank you very much, I'll replace it wit a 74AHCT125. For some reasons 3 of these 126ers are in the box with the 125ers. Don't no were thes come from.

I'll report back once replaced.

ChristophA, 2021-03-28:
All good now! Thanks again for pointing me in the right direction!

closed 2021-03-28 by ChristophA