srg74/WLED-wemos-shield #40: WS2815 with two Data-Lines, any possibility to use both?

tugsi, 2021-03-28:
Hi,
first of  all thanks for your Shield.
Ive used one with a WS2812-Stripe, perfect.
But now I build one for outdoor and want use it with WS2815 12VDC.
my first test works fine, but there is still one question, nowhere found a solution in your Wiki or anywhere else.
The WS2815 works with two DIs, one is for StandBy, if anything happens with the first one.
Was there a way to connect and use the DI2?

Best regards
Thomas

srg74, 2021-03-28:
You have 2 data lines one is marked DI and second is BI. According to data sheet of WS2815 DI is digital input and BI is backup, at first pixel it have to be tight to Ground (GND)

tugsi, 2021-03-28:
Thanks for the answer.
I've forgotten a not entirely unimportant thing, I still have the v1.0 design.
There is no BI, only CLK.

srg74, 2021-03-28:
No better which version you using they all backward compatible with software and strips. BI wire from strip have to be connected to GND not CLK. CLK is a clock pin for APA102 strips. And with newest beta firmware you can use 2 strips per controller even v1.0 :) Just make sure all strips together do not exceed 10A current.

tugsi, 2021-03-28:
Ah, ok, now I understand ;)
I can use 2 stripes?
At the moment I have hung the two stripes one after the other for testing (600 LEDS).
Works well, even without an additional feed of the second stripe.
In the final setting, I probably feed the second stripe in separately so that everything doesn't run over the first and it gets too warm.

closed 2021-04-01 by srg74