srg74/WLED-wemos-shield #44: Support for multiple LED outputs

CWempe, 2021-05-04:
I just read the changelog of [WLED version 0.12.0](https://github.com/Aircoookie/WLED/releases/tag/v0.12.0).
It supports multiple LED outputs.

I guess you could still use just one relay to power on/off all led strips at once.

The level shifter (SN74AHCT125N) is already able to support up to four outputs, too.

Only the additional terminal blocks would need more space, I think.
But maybe we could keep using one `GND` and one `+` for all strips and only add the extra `DI` (and `CLK`?).

I really would like to use a shield with support for at least two LED outputs. 😎

srg74, 2021-05-04:
You can use both DI and CLK as data pins. If you using board v2 also you can set soldering jumpers for 2 more outputs. All available outs shifted. Only downside In this case you can’t use digital microphone.

CWempe, 2021-05-04:
> All available outs shifted.

Great! I did not expect that.

In that case I do not need a new design. 👍

closed 2021-05-04 by CWempe