srg74/WLED-wemos-shield #45: Improve README

CWempe, 2021-05-04:
It has been a long time since I setup this shiled with WLED . :)
So took me a few minutes to get it working again this time.

My issues were:

* I was not sure what pins are used by this shield.
*   I guessed correct it uses the [default pins](https://github.com/Aircoookie/WLED/wiki#default-gpio-usage)
* I forgot to set the relay pin. So the LEDs stayed off at first.

Maybe this information should be added to the README.

srg74, 2021-05-04:
Actually all pins are described in schematic. Pinout available for esp8266 and esp32 boards.

CWempe, 2021-05-04:
I knew I could look for the pins in the schematic or the pcb design.

I just think it would make it easier for the average user if they were mentioned in the README.
Or maybe a short sentence with a link to the schematic.

just my 2 cents 😉

srg74, 2021-05-04:
You absolutely correct. For average user it have to be pointed. Project in constant development and better more clearer navigation in plans.

closed 2021-05-05 by srg74