srg74/WLED-wemos-shield #46: FAQ about maximum load is unclear

CWempe, 2021-05-08:
> The maximum load you can connect is 10A. With greater loads traces leading to and from relay contacts have to be soldered with access of solder to make them ticker due to higher current.

So I can use more than 10 A if I add solder to the traces?!
I just noticed the recommended relay `SRD-05VDCSL-C` has printed `10 A` in the top.

[image]

1. So I guess even with bigger "with access of solder" you cannot/should not use more than 10 A.
2. If the traces (with more solder) can handle more load: Is there a compatible relay available that supports more load (e.g. 15 or 20 A)? Or should this be avoided for this shield anyway?

Maybe the FAQ can be more clear about this.

srg74, 2021-05-08:
I would say it's safe to run 10A with default relay. Actually if you want to run more then 10A you can connect of board relay with greater ratings and proper wiring. One of the user used 15A relay substitute, but I think it's absolute limit because traces on board can't withstand more current.
You can try to find replacement relay by comparing relay dimensions and pin distances. Everything is available in data sheets.

CWempe, 2021-05-08:
> You can try to find replacement relay by comparing relay dimensions and pin distances. Everything is available in data sheets.

Yes. But maybe you already knew one. 😉 
But I think 10A will be sufficient for my projects.

I also noticed the terminal blocks are rated for a maximum load of 13,5 A.

So I would suggest to explicitly not recommend more than 10 A in the FAQ.

Like:
> The maximum load you can connect is 10 A, because the relay only supports 10 A.
>
> If you want to connect a higher load anyway you do this at your own risk!
> Make sure to add a lot of solder to the traces leading to and from relay contacts to make them ticker due to higher current.
> You also need a better relay.
> And be advised that the terminal blocks recommended in the [BOM](https://github.com/srg74/WLED-wemos-shield/blob/master/resources/BOM.pdf) are rated for a maximum of 13,5 A. There are other brands that might support up to 15 A.

srg74, 2021-05-08:
Wiki updated.

closed 2021-05-09 by srg74

zurkes, 2021-05-14:
if you want to high loud look it https://wiki.faked.org/DIY_Ambilight