srg74/WLED-wemos-shield #47: 12v Support

florianmulatz, 2021-05-18:
Hi,

i've ordered these ones from aliexpress (https://www.aliexpress.com/snapshot/0.html?spm=a2g0s.buyer_leave_review.0.0.1fbe6c1bQkU1Ve&orderId=8130740139745887&productId=32848919879) in 10µF (the only "technical" difference from my point of perspective is that they are suitable for up to 50V) for 12V Compatibility - they does not fit perfectly but it's acceptable. I guess these should also work as C4, C5 right.

I'm asking because after I have soldered them and put the Jumper to 12V and added a 12V PSU - the esp does not light up anymore.

Maybe it's a misunderstanding and the 12V Support is only supported for OUTPUT.

Cheers

shortbloke, 2021-05-18:
Did you also install PS1 a VXO7805-500 DC to DC converter to provide 5V to your MCU when using a 12V PSU and 12V LEDs. The AliExpress link doesn't work for me, so I can't see what caps you used for C4 and C5. Also ensure you mount the VXO7805-500 the correct way round.

florianmulatz, 2021-05-18:
ah - that looks to be the culprit! I've completely overseen that component. this should be the correct link for the caps --> https://de.aliexpress.com/item/32848919879.html?spm=a2g0s.12269583.0.0.7ad86a4aEK2JGo. Hm would be challenging to add the dc-dc converter as well because of the too big caps now.

shortbloke, 2021-05-18:
The DC-DC converter does come with 3 right angled header pins, so it can be elevated off the board a little which might provide enough clearance for your larger capacitor pair.

florianmulatz, 2021-05-18:
Sounds legit. Do you think the caps will fit? - thx for your  help indeed!

srg74, 2021-05-18:
> Sounds legit. Do you think the caps will fit? - thx for your help indeed!

Please, for troubleshooting come to WLED discord and we can resolve it quickly.

florianmulatz, 2021-05-20:
connected :) - thx in the meantime

closed 2021-05-20 by florianmulatz