srg74/WLED-wemos-shield #50: Setting leds not work

sagaardu, 2021-06-28:
Hello,

I am trying to test this version but it is impossible to modify the number of LEDs in the configuration.

Can you take a look at it?

Greetings

srg74, 2021-06-28:
> Hello,
> 
> I am trying to test this version but it is impossible to modify the number of LEDs in the configuration.
> 
> Can you take a look at it?
> 
> Greetings

sagaardu, 2021-06-28:
The version I'm using is your compilation to test the usermod via GUI (0.12.2-bl4 "Ryujin")

As you well indicate, you have to configure the number of total LEDs per pin, that value does not allow modifying it, it is at 30 LEDs and it does not allow changing it.

Neither can you modify the usermod presets such as segment-delay-ms: and on-time-s: when you give it to save it leaves them as is

srg74, 2021-06-28:
Please, use official firmware. The one you try is dev fork. https://github.com/srg74/WLED-wemos-shield/tree/master/resources/Firmware/WLED_wemos_shield/v0.12.0_2106271

sagaardu, 2021-06-28:
Hello, I have installed the version you indicate and the problem of configuring the LEDs has been solved.

What still doesn't work is being able to change parameters from the staircase:

- delay-ms parameters:
- on-time-s:

huggy-d1, 2021-06-28:
Are you sending the parameters via browser like the readme.md for the staircase usermod suggests?

sagaardu, 2021-06-28:
No, I am trying to adjust it from the GUI

sagaardu, 2021-06-28:
via curl not working either

` curl -X POST -H "Content-Type: application/json" -d '{"staircase":{"segment-delay-ms":1000,"on-time":10}}' 192.168.100.63/json/state
{"success":true}
`
{"success":true}

srg74, 2021-06-28:
For Json you have to ask in #dev-general at WLED discord channel. They have updated some code related to Json.

closed 2021-07-04 by srg74