srg74/WLED-wemos-shield #51: What's about the possibility to use more than one stripe with a single esp32

Aussiedler, 2021-07-07:
Hi, I have a question about your board. If I understood correctly, the power supply for the board itself can be selected up to 24V. LED strips there are but only 5-12V right? 

I was looking for a variant with an ESP32, because with this it should be possible to connect 10 strips simultaneously and 4 strips productively. Well your board actually fulfills all wishes. But with you I see only the connection option for one strip, or am I mistaken here?

Thanks for the help!

Translated with www.DeepL.com/Translator (free version)

srg74, 2021-07-07:
Hi. Yes you understood correct. Board able to handle 5V or 12/24V power supply and strip must be same voltage. Means 5V PS and 5V LED strip or 12V power supply and 12V LEDs. Regarding quantity of the connected strips 4 strips maximum if you configure regarding your setup. In this case sound reactive feature with microphone is not possible. But possible with linear audio input. Actually quantity of the strips is not beneficial only make sense if your setup is required connections in one point.

Aussiedler, 2021-07-07:
Thanks for the quick feedback! In my case I have a large room with 2 supply points where 5x25mm² cable. A total of 30m LED strips must be connected. I had considered using 2 WLED controllers which control 2 lines each. So I would have per line 7-8m or 420-480 LED pixels per line. The ESP32 should be able to handle that.

How is another stripe connected to your board?

srg74, 2021-07-07:
Here is pinout for entire board https://github.com/srg74/WLED-wemos-shield/blob/master/resources/Images/wiki/Wemos_shield_pinout.pdf

closed 2021-07-09 by srg74