srg74/WLED-wemos-shield #52: Thoughts about Assembly-Wiki

CWempe, 2021-08-01:
I ordered a few v2.0. board a while ago and am going to solder them now.
Lucky me, v2.1 has been released since then. 😅

Anyway.
I will try to follow the [Assembly-Wiki](https://github.com/srg74/WLED-wemos-shield/wiki/Assembly).
But here are already a few thought after reading the guide.

1. I think it would be nice if the parts that are only for v2.1 were marked accordingly.  
  But maybe there are not many people who have the older version 2.0.
2. Chapter 1: `IO15 - Let you choose if your output 15 is shifted or not (soldered left not shifted);`  
   a. I guess `output 15` is a pin on ESP32? So I can ignore this for my ESP8266?  
   b. You mean: "Let's you choose/define if your output 15 should be shifted or not". Right?
3. Chapter 1: `IO15-1 - Must be soldered if output 15 is shifted;`  
  "Must be soldered if IO15 is soldered, too." But I am not quiet should sure if this is what you mean.
4. Capter 1: `J-R2` and `J-R1`
  Would be nice if you would linke to the [option](https://github.com/srg74/WLED-wemos-shield/wiki/Options) this is related to.  
  As a "dumb user" I am not sure if I need there resistors. At least not without further research of the wiki.
5. Chapter 2: `Check orientation with markings on PCB.`
  I would indent this line as a sub item from the Diodes line above.
6. Chapter 7: The image does not work.  
  [image]

Thank you anyway for the Assembly-Guide! 👍

srg74, 2021-08-01:
> I ordered a few v2.0. board a while ago and am going to solder them now.
> Lucky me, v2.1 has been released since then. 😅
>
> Anyway.
> I will try to follow the [...]

All parts remain unchanged. No major changes where performed.

> 2. Chapter 1: `IO15 - Let you choose if your output 15 is shifted or not (soldered left not shifted);`
>    a. I guess `output 15` is a pin on ESP32? So I can ignore this for my ESP8266?
>    b. You mean: "Let's you choose/define if your output 15 should be shifted or not". Right?

Additional shifted outputs only for ESP32

> 3. Chapter 1: `IO15-1 - Must be soldered if output 15 is shifted;`
>    "Must be soldered if IO15 is soldered, too." But I am not quiet should sure if this is what you mean.

Yes, if you shifting output jumper with PIN-1 must be soldered too.

> 4. Capter 1: `J-R2` and `J-R1`
>    Would be nice if you would linke to the [option](https://github.com/srg74/WLED-wemos-shield/wiki/Options) this is related to.

There is explanation for those. R2 is only enabled if you have a old LED strips. R1 only enabled if your board is not stable without pull-up resistor.

>    As a "dumb user" I am not sure if I need there resistors. At least not without further research of the wiki.

There is not such a thing as dumb user. Everything take a learning curve. Any electronic project is not a toy.

> 5. Chapter 2: `Check orientation with markings on PCB.`
>    I would indent this line as a sub item from the Diodes line above.

Fixed, for more clear understanding.

> 6. Chapter 7: The image does not work.
>    [image]

Fixed

> 7. Chapter 2: I think you forget to mention R2. It is soldered in the image.

Read ["Options"](https://github.com/srg74/WLED-wemos-shield/wiki/Options) first

CWempe, 2021-08-02:
> All parts remain unchanged. No major changes where performed.

I literally stumbled on the first line.
I cannot find any `IO1-IO3` an my v2.0 pcb.
But this might be the only change. 😄 

> There is explanation for those. R2 is only enabled if you have a old LED strips. R1 only enabled if your board is not stable without pull-up resistor.

I just think it would be easier for the user if you link to these explanations.

> Fixed

There are no red traces in the image. 😉 
[image]

CWempe, 2021-08-02:
And I just found out the `TTGO T7 V1.3 MINI 32 ESP32` does not really fit. 😕

I gues I will get it to work.

But maybe this should be mentioned in the assembly guide.
The user could insert the ESP32 before soldering the relay.
It would fit if the relay would be slightly pushed to terminal connectors or angled a little.

srg74, 2021-08-02:
Actually it's a manufacturer problem. They have a production with not consistent board size. When I add this board to Wiki USB port was 0.5mm from relay, with relay in correct position.

CWempe, 2021-08-02:
> Actually it's a manufacturer problem. They have a production with not consistent board size. When I add this board to Wiki USB port was 0.5mm from relay, with relay in correct position.

I guess that is the downside of a low priced DIY project. 😄

srg74, 2021-08-02:
I would not say downside. It's a manufacturer who really don't care about customers. My boards working in 1000's quantities and most of the time with no problem. Shield was tested very vigorously by many users and recent version have no complains. Also if you don't like DIY there is fully assembled and individually tested version available.

CWempe, 2021-08-02:
I did not mean your project is bad.
I meant chinese manufacturers are sloppy.

If we would use expensive parts for our projects this would not be an issue.

You are doing a great job with this project!. 👍

srg74, 2021-08-03:
No offense taken. I'm just trying to explain how project was evolving. Parts gotta be a good quality but not expensive. That was a goal.

closed 2021-08-25 by srg74