srg74/WLED-wemos-shield #53: Sound reactive build 2108010 bricks ESP32

lanman1, 2021-08-03:
OTA build 2108010 bricked ESP32 using both with/without display.  Had to re-flash with USB

srg74, 2021-08-03:
I will check firmware and if this fixable I will update files

lanman1, 2021-08-03:
Thanks!  No problem, just letting you know.

srg74, 2021-08-03:
Fixed :)

lanman1, 2021-08-03:
Must be me, When I try to do an OTA update of my TTGO ESP32, it bricks with
the predone bin.  I then have to reflash with USB.

srg74, 2021-08-03:
Again? With new bin files?

lanman1, 2021-08-03:
yes.  I have run 13. on this before and it is ok.  If I'm on 12.0, I upload the bin via OTA and the ESP just lock up at reboot.  No relay click, just locks.

lanman1, 2021-08-03:
The non-SoundReactive v0.13.0-b2/esp32_no_mods.bin works ok.  But both SoundReactive bin do not.

srg74, 2021-08-03:
TTGO ESP32 T7 v1.3?

lanman1, 2021-08-03:
Yes.  As information, it appears to be related to presets.  I did a factory reset and was able to load the soundreactive bin ok but there is no option to backup/restore presets.  I flashed back to the non SoundReactive and preset backup/restore is there.

srg74, 2021-08-03:
So you did fix your board and bin files is ok? Presets backup your_wled_IP/edit

lanman1, 2021-08-03:
There seems to be an issue with soundreactive bin files.  There is on backup/restore option in the soundreactive bins and if presets exist before flashing soundreactive, the board locks up.

srg74, 2021-08-03:
Yep, I can confirm. In this case have to report to sound reactive fork.

lanman1, 2021-08-03:
Thank you.  Let me know if I can help test, thank you for posting the .bin(s)

srg74, 2021-08-03:
I already posted question to devs. If glitch will be fixed I'll post result here.

srg74, 2021-08-04:
After some clarification I found that bin files is OK. To back up presets. you go to IP_wled/edit -> right click on file presets.json and save file.

lanman1, 2021-08-04:
Ok.  There still is the issue of if presets exist during ota upgrade it
will brick the board.

srg74, 2021-08-04:
I will dig further. Hopefully there is explanation to this issue.

srg74, 2021-09-16:
Have been fixed.

closed 2021-09-16 by srg74