srg74/WLED-wemos-shield #54: Firmware and D2 Problem

Leon751, 2021-08-20:
Hi there,
thank you for your great work.
I assembled your board V2.1.
It works but i have some problems with it.

1. I downloaded and installed your firmware.
The firmware sound_reactive : SR_shield_esp32_display_dallas.bin and : SR_shield_esp32_display_no_dallas.bin
is not working fine.
It stutters every few seconds.
The firmware SR_esp32dev.bin  run without problems.
i have installed the temperature sensor dallas too so it is not usable.

2. The next problem is when I connect another LED strip to D2 and configure it, then white dots flicker  sometimes.
I think that's too much for the EPS32. 
I built 2 lamps with 230LED WS2815 each. That will be a little too much for the ESP32 or not? 
The system runs very well without D2. 
I have now connected both lamps to D1 and that works without any problems, but only synchronously

3. For what is the relay? If I switch the LED on or off, they do it before the relay switches. The relay switches with a delay, then the LEDs are already on.

Excuse my english. I used a translator to help.

Greetings Erwin W.

srg74, 2021-08-20:
Thank you for good words. 
1) Regarding Dallas sensor. I have to check firmware. Maybe there’s a new bug introduced. 
2) D2 output is configurable you have solder jumper IO1-IO3. Means I one position you choose GPIO1 as out and second is GPIO3. Try to switch outputs. Quantity of LEDs is ok to drive. 
3) I don’t see how you connect your LEDs but relay cutting of power to strip. Yes it’s working with a delay for a reason.
You don’t have to apologize for your language. Our community is international and it’s expected.

Leon751, 2021-09-01:
Hi there,
I've tried a few things.
Now I also have an oled display on it.
The relay works.
Solder jumper IO1-IO3 was set.
It's OK
I haven't tried D2 any more.
The firmware runs without an oled display.
The firmware with OLED display and dallas sensor has errors.
It appears in the app
"TypeError: Cannot read property` min´ of undefined.
No segments can be defined either.
And the led stutter.
Alexa is not activated in all firmwares that I have tried.
Can't you provide the source code for download so that Alexa or IR can be activated?

srg74, 2021-09-01:
It's a lot of questions to answer at once. If you want to troubleshoot please visit WLED Discord channel. I will be there and gladly help you.

Leon751, 2021-09-02:
I am sorry for all of these questions.
I have one request, can you change the version of the mini_shield and upload as bin?
This runs very well and the display is already activated.

wemos_shield_mini_esp32_display_and_sound.bin

V0.12.0 "Kuuhaku"

Activate alexa in the weld.h folder

// # define WLED_DISABLE_ALEXA // saves 11kb

 once alexa and infrared

// # define WLED_DISABLE_ALEXA // saves 11kb
// # define WLED_DISABLE_INFRARED // there is no pin left for this on ESP8266-01, saves 12kb

Maybe all this again with the dallas temperature sensor?

That would be really nice and I hope that this version works well with alexa and more
Thank you

srg74, 2021-09-02:
Hi! You don’t have to be sorry. Reason why I ask you to join WLED Discord because it’s easier to collaborate and solve issues in real time. If question is not simple we will make a long tread and there’s a chance that issue will not be solved. V0.12 of WLED firmware is a little buggy. Better to use v0.13-b2, a lot of bug fixes was done.

Leon751, 2021-09-02:
is that this page?
https://wled.discourse.group/t/welcome-to-the-new-wled-community-should-you-post-here-or-open-an-issue/22
isn't it  only for the original wled?
I need the sound reactive wled for your board.
version V0.12. runs better than v0.13-b2

srg74, 2021-09-02:
https://discord.com/invite/KuqP7NE

closed 2021-09-16 by srg74