srg74/WLED-wemos-shield #55: Question: WS2801 on v2.1 board

habbo656, 2021-09-02:
Hi,

thanks for that nice board!

I want to connect a WS2801 stripe to the bard - is it possible? What di I need to solder to have GPIO0 on D2 output?

Currently I have he bridge IO1-IO3 bridged for IO1 - need that bridge to be open? Looking at the pinout for v2.1 it does not look like I have the option to use GPIO0 as CLK but with WLED I can just use GPIO2 and GPIO0 for the WS2801.

Best regards,
Thomas

srg74, 2021-09-02:
There is a 2 pins exposed at the time. GPIO2 for DI and GPIO3 for CLK. In case you want to use second terminal output you have to short solder jumper in needed position. Middle pad and one of your choice IO1 or IO3. GPIO0 is reserved for push button, because it have special function to reset controller to default.

srg74, 2021-09-16:
Answered

closed 2021-09-16 by srg74