srg74/WLED-wemos-shield #57: Question about the solder jumpers + typo in the documentation?

nreymundo, 2021-09-30:
Hey! got a couple questions/doubts about the jumpers that need to be soldered that I can't figure out or find the right answer. Hope this is the proper place to ask but feel free to point me in a direction and RTFM :P. 

```
IO15 (ESP32 only)- Let you choose if your output 15 is shifted or not (soldered left not shifted);
IO15-1 (ESP32 only) - Must be soldered if output 15 is shifted;
IO14 (ESP32 only) - Let you choose if your output 14 is shifted or not (soldered right not shifted);
IO14-1 (ESP32 only) - Must be soldered if output 14 is shifted;
```
I'm using  D1 Mini with a ESP8266 so I'm assuming I can ignore all the ESP32 only jumpers and leave them as is. 

```
J-R2 - If soldered bypassing resistor R2;
```
From what I understand R2 is not needed for W8211B but no harm in using them, right? In any case is either soldering the resistor or the jumper. 

```
IO1-IO3 - Let you choose D2 output, GPIO1 or GPIO3 (Feature added in shield v2.1);
```
This is the one I have no idea about. Following the pictures in the assembly guide I'm guessing I need to solder left for GPIO1 but couldn't find any documentation or don't know what to look for. 

```
J-R1 - Soldered if resistor R1 must be used;
```
The FAQ section on "[Do I need R1?](https://github.com/srg74/WLED-wemos-shield/wiki/FAQ#do-i-need-r1)" mentions to solder `JP3` if using the button but I think it should be `J-R1`?. 

Thanks!.

srg74, 2021-09-30:
> I'm using D1 Mini with a ESP8266 so I'm assuming I can ignore all the ESP32 only jumpers and leave them as is.
> 
Yes disregard ESP32 part.
> ```
> J-R2 - If soldered bypassing resistor R2;
> ```
> From what I understand R2 is not needed for W8211B but no harm in using them, right? In any case is either soldering the resistor or the jumper.
> 
No need for resistor just solder jumper.
> ```
> IO1-IO3 - Let you choose D2 output, GPIO1 or GPIO3 (Feature added in shield v2.1);
> ```
> 
> This is the one I have no idea about. Following the pictures in the assembly guide I'm guessing I need to solder left for GPIO1 but couldn't find any documentation or don't know what to look for.
> 
If you using second strip connected to terminal D2 then you have to choose one of the GPIO 3 or 1.

> ```
> J-R1 - Soldered if resistor R1 must be used;
> ```
> 
> The FAQ section on "[Do I need R1?](https://github.com/srg74/WLED-wemos-shield/wiki/FAQ#do-i-need-r1)" mentions to solder `JP3` if using the button but I think it should be `J-R1`?.
> 
Yes it's a error left behind from previous version of the guide. Corrected for J-R1

Thank you for your post

nreymundo, 2021-09-30:
> If you using second strip connected to terminal D2 then you have to choose one of the GPIO 3 or 1.

Ah awesome!. Just so happen that I was eventually going to try to figure out how to make 2 strips work with the same shield or if that was even possible haha. Is there any documentation or anything I could read to decide which GPIO to use or would either work for a D1 Mini?. 

Anyway, thanks so much for the quick reply!.

srg74, 2021-09-30:
GPIO 3 is hardware accelerated. With recent firmware version there is no problem to drive to separate strips. ESP8266 can drive from 2 pins easy. More info about firmware you can read at https://kno.wled.ge

nreymundo, 2021-09-30:
Great! Thanks for the quick help.

closed 2021-09-30 by nreymundo