srg74/WLED-wemos-shield #58: Multi Relay setup

siedi, 2021-11-07:
Hi,

in the documentation, you mention a multi relay setup. Actually, there are pins for the power supply (5V) for a second (external) relay. But I can't find the pin to connect to the D6/GPIO12 as the internal relay is connected to. 
Is that possible at all, or do I need to use one of the free external GPIOs and change the wled software?

-Siedi

srg74, 2021-11-07:
External relay board can be connected to any available exposed pin. https://github.com/srg74/WLED-wemos-shield/blob/master/resources/Images/wiki/Wemos_shield_pinout.pdf Just be aware that you need to use external relay board and not a relay only due to high current draw by relay coil.

blark, 2021-11-30:
It would be nice if you could add some documentation around using a second relay, along with a separate digital output for controlling additional pixels. Thanks.

srg74, 2021-11-30:
As I mentioned above there is pinout diagram. According to that pinout output D2/CLK could be used as second output for LEDs. Additional Relay board can be connected to any exposed pins on add-on header. Numbers of those pins on pinout diagram.

blark, 2021-12-01:
It would be very nice to have some practical examples in addition to the pinout diagram (ps. I purchased two of your boards from Tindie - great work). 

I have designed and manufactured my own boards with ESP32s on them, I consider myself somewhat capable in this area... I could examine the schematic, read the data sheet for the logic level shifter, to be sure I understand things and am doing this correctly. But it would be great to have the benefit of easy hook-up instructions to save that time. Thanks for your time and effort here!

srg74, 2021-12-01:
I got your point. Need some time to figure out how to make guides understandable by most of the users.
Thanks for a comment.

blark, 2021-12-02:
Great! I was checking out the Quinled project last night and there are some decent [wiring guides there](https://quinled.info/quinled-dig-uno-wiring-guide/) that you might want to check out. Although I like your board design with the built in relay better ;)

srg74, 2021-12-14:
Closing due to no activity.

closed 2021-12-14 by srg74