srg74/WLED-wemos-shield #61: experimental/Firmware/esp32_display.bin build 2111281

o1hitman1o, 2021-11-29:
experimental/Firmware/esp32_display.bin The display is not functioning. But in experimental/Firmware/Debug_builds/esp32_display.bin Everything is working correctly.
Not sure what the difference is.

srg74, 2021-11-29:
There is no difference in builds, only debug build is debug enabled.

o1hitman1o, 2021-11-30:
I have tested every build from v0.11.1_210210 on v1.51 board to the latest version. This was the results.

wemos_shield_esp8266.bin v0.11.1_2102210 - working with display
wemos_shield_esp8266.bin v0.11.1_2102130 - working with display
wemos_shield_esp8266_display.bin v0.11.1_2102130 Update to recent - working no display
esp8266_display.bin v0.12.0_2106280 working no display
esp8266_display.bin v0.13.0-b2 - working no display
esp8266_display.bin v0.13.0-b3 - working no display
esp8266_display.bin v0.13.0-b4 - won't boot
esp8266_display.bin v0.13.0-b5 - won't boot
esp8266_display.bin experimental - working no display
esp8266_display.bin experimental debug - working with display
 I have tried flashing several times and get the same result.
There has to be some difference somewhere I just don't know where to look.

closed 2021-12-10 by o1hitman1o