srg74/WLED-wemos-shield #65: Level issues?

drapergeek, 2022-01-29:
Hi!

I ordered a shield from Tindie to try and offset some issues that I believed were based on the need for a logic level shifter. I have a d1 mini with WLED powering 3 short LED strips. The strips are mounted a couple feet from where the controller is mounted. They work great, except that if the lights are "off", when we turn on a fan or a humidifer, we end up with a few of the LEDS turning on at the beginning of the strip. As I mentioned before, my searching lead me to believe this was due to the fact that my strips were mounted away from the controller and I needed a level shifter. 

I ordered and installed your board and sadly, I'm still experiencing the same issues. I made sure to wire the power in and the power out as your diagrams show. Is there something else I could be missing? Sorry for the beginner question but I'm at a lose for my next move!

srg74, 2022-01-29:
Could you please move discussion to WLED Discord?

drapergeek, 2022-01-29:
No thanks. I thought since I purchased your part, you'd be able to assist.

srg74, 2022-01-29:
You got me wrong. I did not reject you to help. Just WLED Discord allow me to collaborate about your issue in person. I'm on Discord @Serg74 waiting for you. You can DM me or talk in public.

drapergeek, 2022-01-29:
My apologies for the misunderstanding. I'll reach out in the Discord in the future. Thank you!

srg74, 2022-02-22:
solved

closed 2022-02-22 by srg74