srg74/WLED-wemos-shield #66: soundReactive_d1_mini.bin / v0.13.0-b3 / No Alexa ???

bennony, 2022-02-01:
soundReactive_d1_mini.bin / v0.13.0-b3

Hi, everything works quite well so far, but I can't get the ESP connected to Alexa. The checkmark for Emulate Alexa device is set. Restarting Alexa and searching again had no success either

srg74, 2022-02-02:
All questions related to functionality of particular firmware have to be addressed to developers. I'm only compiling code and publish to my repository for users convenience.
You can try latest code here https://github.com/srg74/WLED-wemos-shield/tree/master/resources/Firmware/Sound_reactive/v0.13.0-b6

bennony, 2022-02-09:
I tested the b6 (aircoookie) (ESP 8266 support) Alexa works, the sound-Reactive B6 unfortunately does not support 8266, the 13.0 b4 was the last one here (however, alexa seems to be deactivated here in the script) I would be happy if you still have the b6 finished for the ESP8266 (with Alexa Enabled ;-)) the sound reactive is a fine thing.

srg74, 2022-02-09:
Next update of sound reactive to b6 is not depend on me but on SR development team. As soon they release new update on ESP8266 fork I will compile and post new bins.

bennony, 2022-02-09:
Great news ;-) and great work

closed 2022-02-22 by srg74