srg74/WLED-wemos-shield #70: Help with part selection at LCSC

timtom2k, 2022-04-10:
Hi everyone, 

it took me a while to update the BOM_Digi-key_v3.0.csv to LCSC with the help of the BOM_v2.1_LCSC.xlsx as parts are much less expensive as most other sources.

As I'm not an electronic expert, I'm comparing the digi-key specs with LCSC and struggeling with the following parts:
1) C4, C5 and IR: C162684 and C141632 should be selected. However, compared to other capacitors and IR the price seems a bit higher. Are there good equivalents at LCSC that can be used?
2) C3 0.01uF: I found several capacitors at LCSC, but don't know which one can be used?
3) Fuse holder is not available at LCSC, correct?

Best regards,
tim

srg74, 2022-04-10:
There is not a lot of alternatives for parts you mentioned. C3 could be C2832505
C4, C5 is only C162684 in stock
For IR sensor I would recommend only sensor being tested with WLED.
Fuse holder can be sourced from different suppliers.

closed 2022-04-17 by srg74

xrtze, 2022-08-12:
@timtom2k Any chance you can provide the updated BOM for LSCS?