srg74/WLED-wemos-shield #73: Please publish the build environments for the firmware files

microraptor, 2022-08-13:
It would be helpful if the build environments for the firmwares under /resources/Firmware/ and /resources/experimental/Firmware/ were published. This would help people, who want a mix of features that isn't compiled or just do generally learn what the environments should look like. As a quick example the file listing the environments could look something like this:

```
...

File:
/resources/experimental/Firmware/WLED_0.14.0-bl0_ESP32_DIG.bin

Source:
https://github.com/blazoncek/WLED/tree/segment-api

Environment:
[env:quin_mini32]
extends = env:esp32dev
platform = ${esp32.platform} ;or espressif32@3.5.0
board_build.partitions = tools/WLED_ESP32_4MB_1MB_FS.csv
board_build.filesystem = littlefs
build_unflags = ${common.build_unflags}
build_flags = ${common.build_flags_esp32} -D WLED_RELEASE_NAME=ESP32_DIG
  -D USERMOD_DALLASTEMPERATURE
  -D TEMPERATURE_PIN=13
  -D LEDPIN=16
  -D DATA_PINS=16,3,1,4
  -D RLYPIN=15
  -D BTNPIN=0
  -D USERMOD_PIRSWITCH
  -D PIR_SENSOR_PIN=32    # Q4
  -D USERMOD_MULTI_RELAY  # Q1 (15), Q2 (12)
  -D MULTI_RELAY_MAX_RELAYS=2
  -UWLED_USE_MY_CONFIG
lib_deps = ${esp32.lib_deps}
  paulstoffregen/OneWire@~2.3.6

...
```

srg74, 2022-08-14:
All builds are self explanatory. For example build from WLED-wemos-shield/resources/Firmware/WLED_wemos_shield/v0.13.2-a0/ is original build from @Aircookie repository, file named WLED_0.13.2-a0_SP501E_2m.bin means v0.13.2-a0 build for SP501E controller.

microraptor, 2022-08-14:
Well it would be mostly for the build environments, because they are often a little more difficult to figure out with usermods. Since you have done so many builds I was hoping you already have a file laying around with all the environment variations.

srg74, 2022-08-14:
I posted an example which mostly reflect most of the builds.

microraptor, 2022-08-14:
That looks great. Thanks a lot :)

closed 2022-08-14 by microraptor