srg74/WLED-wemos-shield #75: What is the relay for?

Eddiiie, 2022-09-03:
Question here, what exactly is the relay for?   I have several power supplies along the run..   Is the relay to turn on power to the power supplies before starting the LEDs (sending data)?  Allowing some time for them to come up to power, or whatever..
Sorry if I missed this in the documents somewhere.  :)

srg74, 2022-09-03:
Relay is for disconnecting power for LEDs strip. First it’s for energy saving, LEDs consuming energy when turn “off”. Plus it’s extending LEDs life time.

Eddiiie, 2022-09-03:
"Relay is for disconnecting power for LEDs strip."

So I would take the + and GND coming off the shield  and connect that to another relay that is connected to the power supplies?

srg74, 2022-09-03:
Shield can handle up to 10A. You connect power supply to shield and LEDs to output on the shield, all wires. Here is pinout https://github.com/srg74/WLED-wemos-shield/wiki/Pinout

Eddiiie, 2022-09-03:
I have 1 WLED controller, 1200+ LEDs in a single row that run on top of my fence, 4 power supplies all chained on an extension cord.   For this, I would add a 5th power supply to power the ESP32 and external relay (on all the time), run wires from LED+ and GND from the shield to power the external relay that is in turn connected to the AC to the LED power supplies?
I think this will work.

srg74, 2022-09-04:
Board is rated only 10A. If you draw more than 10A you can turn on and off secondary relays from board. All LEDs must be connected to secondary relays and fused. Or you can bypass relay and connect without relays. All depends on your setup. It’s difficult to say without knowing your setup.

closed 2022-09-15 by srg74