srg74/WLED-wemos-shield #76: Could you help me on the pinout ?

Roxcreed, 2022-09-22:
Hello

First of all, thanks for your schematic and your guide
I'm having a hard time trying to find why my LEDs doesn't get neither Data nor Clock. I have a led strip with 4 wire (so 2 for power, 1 data, 1 clock). The relay is working good (I measure the output with my voltmeter, there is 5V, so no issue here). But on D1 , D2.CL I have nothing

My controler is an ESP32 from quinled. I have used this pinout : https://github.com/srg74/WLED-wemos-shield/wiki/Pinout to set mine. 
[image]

I03-I01 has solder on the right side. With this setup my LEDs stay black. (I try to change the clock for GPIO 1 or 3 and vice versa, nothing happened)

But if I use the wire to directly touch D4 for data and TX for clock ... it works. Any idea why I don't get signal on terminals ?

srg74, 2022-09-22:
I need more details. Did you used my shield? DIY or you bought it? What is your setup? Better reach me on WLED discord.

Roxcreed, 2022-09-22:
Thanks to be still there, I was a bit worried 

I use your shield 3.0 with your PCB. I order according to the BOM. I put all the bare minimum components and the components for the power jumper selector 

So my ESP32 is from quinled (but the board where I plug it, is your shield 3.0). I flash according to your guide. So I use these files : 
- bootloader_dio_40m.bin
- partition_4mb.bin
- WLED_0.13.3_esp32_no_mods.bin

I have a 5V power brick and my LED strip is SK9822 (similar to APA102)

How do I find you on discord ? My ID is Graw#0561

closed 2022-09-24 by Roxcreed