srg74/WLED-wemos-shield #92: Add the ethernet usermod to the autobuild

joggs, 2022-12-04:
Feature request.
Hi!
As many are using the quinleds with built in ethernet now, would it be possible for you to add the ethernet usermod to the autobuild to at least the esp32 max version, eg:  WLEDMM_0.14.x.x_esp32_16MB_max.bin ?
A big thank you for these auto builds!

ewoudwijma, 2022-12-04:
Hi Joggs,

Do you also need -D QUINLED_AN_PENTA in that build?
And -D WLED_USE_ETHERNET?

Best is if you provide a platformio environment you are using now.

And my plan was already to provide setups for quinled boards where all pins are pre-configured. Is that in your platformio environment?
If not, can you provide how pins are used for ethernet, microphone, 4 line display, led gpio's and what not.

joggs, 2022-12-04:
Thanks for your quick response!
For me, i think -D WLED_USE_ETHERNET is enough as I do not use the penta.
Actually I am not familiar with platformio yet, so I have to google that and learn a bit more before I can answer. When it comes to quinled, there are a couple of different esp32 hats you can buy with different types of extra hardware , so I think the creator of quinled, @quindor, could give a better answer than me. as there are different revisions etc.
For my mic setup, I am using a separate I2S mic connected to
Pin I2S Serial Data: 12
Pin I2S L/R Clock: 2
Pin I2S Serial Clock: 32
I earlier used the quinled esp32 hat that have built in mic, which I think use the same ports as above.
Also, there are different ports for different hw revisions I think. Pinout for the quad you can find here: https://quinled.info/quinled-dig-quad-pinout-guide/

srg74, 2022-12-04:
I uploaded bin file https://github.com/srg74/WLED-wemos-shield/blob/master/resources/experimental/Firmware/AudioReactive/WLEDMM_0.14.0.8_wemos_shield_esp32_16MB_max_eth.bin but you have to reconfigure pins for your board.

joggs, 2022-12-05:
Thank you!
I can now see and modify the ethernet type under wifi-settings, however, if
you have selected a module in the ethernet-type, I think that the expected
behavior is that  the wifi connection/module should be disabled, but the
wifi connection is still active and the ethernet connection is not
activated. I suppose this is something that has to be fixed in the source
code and out of this scope?

intermittech, 2022-12-05:
@joggs as long as you are using a QuinLED-ESP32-ABE (Ethernet) those pins and configuration values will always be the same, no matter what board you put it on. That board might however require configuration such as for the [Dig-Uno](https://quinled.info/quinled-dig-uno-pinout-guide/), [Dig-Quad](https://quinled.info/quinled-dig-quad-pinout-guide/) or the [An-Penta](https://quinled.info/quinled-an-penta-diy-pinout-guide/) that was mentioned.

closed 2022-12-10 by srg74

joggs, 2022-12-10:
> @joggs as long as you are using a QuinLED-ESP32-ABE (Ethernet) those pins and configuration values will always be the same, no matter what board you put it on. That board might however require configuration such as for the [Dig-Uno](https://quinled.info/quinled-dig-uno-pinout-guide/), [Dig-Quad](https://quinled.info/quinled-dig-quad-pinout-guide/) or the [An-Penta](https://quinled.info/quinled-an-penta-diy-pinout-guide/) that was mentioned.
> 
> So the standard definition included in WLED should do that perfectly and nothing else to configure for Ethernet then just selecting it.
> 
> In regards to the WiFi bit, what we generally recommend is that you join WLED-AP and from there enable Ethernet, that way it never joins a WiFi SSID.

Yes, the lan worked well with the ordinary WLED release and I used it for some months, but with this build I select the correct lan interface just as I usually do, but the lan does not work after I do this, so there must be something in the backend that is not working for the quinled quad lan option

ewoudwijma, 2022-12-10:
What do you mean with ordinary WLED? WLED 0.13 or latest dev (0.14 beta)?

joggs, 2022-12-10:
Sorry, I meant 0.13. With that release. I just install it, connect with
wifi and sets the quinled lan option and reboots and I have wired
connection.
With this build I do not have a working wired connection after reboot.

ewoudwijma, 2022-12-10:
Okay, try the 0.14 beta Ethernet bin which you can find here:

https://github.com/srg74/WLED-wemos-shield/tree/master/resources/experimental/Firmware

If this is also problematic you could ask for help win WLED net-testing channel on discord

joggs, 2022-12-11:
Yes, the beta works as expected:
(
https://github.com/srg74/WLED-wemos-shield/blob/master/resources/experimental/Firmware/WLED_0.14.0-b0_ESP32_ethernet_usermod.bin
)

But not the audio reactive build, which is the one I would like to use:
(
https://github.com/srg74/WLED-wemos-shield/blob/master/resources/experimental/Firmware/AudioReactive/WLEDMM_0.14.0.10_esp32_16MB_max_eth.bin
)

ewoudwijma, 2022-12-11:
@srg74 , are the plaformio environment settings exactly the same in above both bins?
As I think there is no difference in Ethernet implementation in WLED AC and WLED MM...

joggs, 2022-12-11:
Update: 
I kept the config and presets when flashing between the audio reactive build, to the ethernet build you wanted me to try out and I also kept the config when I flashed back to the audio rective build again, and after that, the ethernet still works on the audio reactive build.

I then just had to try to set the lan to none, to see what happens. The wifi was enabled as expected. I then set LAN to quinled-esp32 again, but then the LAN was not enabled, I was still on wifi.
I haven't played around with power cycling of  the unit up until now as I never had to do this, but now I tried that, and then after bootup, the LAN was enabled. I did not have to do this with the beta.  With the beta, it was enough with the automatic restart that occurs after you change the LAN device and save.
Not sure what you can make of this, but hope it helps.