srg74/WLED-wemos-shield #95: WLED not booting after having no power for a while

DopeCoder01, 2022-12-08:
Hi, I have been having an issue with the shield. I already posted the issue on the WLED discord but didn't get any useful help there.
I'm using the 3.0 board, with an ESP32 (this one to be exact: https://nl.aliexpress.com/item/32858054775.html?spm=a2g0o.order_list.order_list_main.20.56f779d22XlEkK&gatewayAdapt=glo2nld, with a CP2104 module).

The problem I have is reliable and repeats itself every time I do this:
1. insert the ESP32, turn on the PSU;
2. everything works as it should: WLED boots, the relay is turned on and the LEDS light up;
3. When I turn off the power, but turn it on again reasonably fast (within a minute or 2) it works like normal again;
4. But, when I turn off the power for longer than a certain period of time (which I haven't determined yet), and turn it on again, WLED refuses to boot. Power is provided to the ESP, as the onboard red LED is lit up;
5. Nothing I do can restore this. Except most of the time, when I simply remove the ESP from the headers and reinstall it, it works again.
6. Doing all of this with an ESP8266 poses no problems at all.

I have read several other problems regarding WLED not booting, which were due to some strapping pins preventing the boot of WLED. However, in those cases WLED couldn't boot at all, while in my case it does boot reliably at the start.

P.S.
I'm posting this here, because I think this problem is related closely to the hardware configuration. I hope that I'm in the right place.

srg74, 2022-12-08:
I understand your frustration. Design is exist for more than a year and if your esp32 is ok everything should work. Most likely your problem in ESP32 board.

DopeCoder01, 2022-12-08:
Okay, something I already kind of expected, but it's good to hear from someone who has a better understanding of it than me. Thank you!

closed 2022-12-08 by DopeCoder01