srg74/WLED-wemos-shield #106: ESP32 D1 Mini working, no power on output terminal connectors

timmgleason, 2023-04-07:
Going to start simple, as this is the most complex assembly I have yet attempted.
Ordered PCBs, ordered parts (LCSC, Digikey, and on hand), using AITRIP D1 Mini ESP32, have Tindie 16MB version on hand once I get things working.
Passed power check, as per wiki Check page. D1 Mini powers up, flashed bootloader, 4mb partitions, and firmware. That all works fine installed into wemos shield board. Can connect and configure.
Have SK6812 and a WS2812 test strips on hand.
Cannot get either of them to light up after configuring WLED for either/or. Testing shows zero volts on G and + for X1/X2 output terminals, which indicates a problem somewhere between D1 and output terminals.

I'm looking for suggestions as to where to start looking first and what other information might be needed to answer that question (pics, jumpers soldered, etc).

Thanks,
Timm

rstarkey, 2023-04-07:
I just built a couple controllers myself and went through this.  Does the relay click when you power it on?  If not, make sure you have the correct relay GPIO configured in WLED.  It should be 19.

timmgleason, 2023-04-08:
Checking that now. I somehow missed that in all the pages of information....

timmgleason, 2023-04-08:
Nope, no relay click.
I have had the SK6812 lights working on a raw D1 mini using 0.14.0-b1 before, but they do not control especially well near the end of the run. That's why I thought using the wemos shield would be a better option (more power, more options, etc).

rstarkey, 2023-04-08:
Yeah, so go into LED Settings in WLED and enter in the correct GPIO pin for the relay. Save it and reboot.  It should make it come to life.

As for running the strip, depending on how long the strip is, you still may need to do power injection.

timmgleason, 2023-04-08:
Yes, did that (full power removal reboot). Still nothing from the relay. Assuming we are both referring to the Songle SRD (K1).
Power light on D1 mini, power light on Wemos board. 
Using a 10A 5V power supply. Light string I was working with is only 99 lights and works fine with its' original 1.2A power supply, but that controller just does not provide good options.
I am now worried that some component I installed is damaged or some such. Still a relative novice with electronics beyond a breakout board.

rstarkey, 2023-04-08:
Ok, so the ESP32 gets power, so that's good.  We know the power in and fuse is good.  Now you just need to break the problem down into small pieces and test each bit.

Check your soldering work.  Make sure you don't have any bridges, open joints, missed spots, and/or shorts.  

Then I'd use a multimeter to check:
* both sides of each diode to make sure you see voltage.  If you don't, it's reversed.
* Double check to make sure you didn't miss any components (I forgot a cap on one of my boards) 
* probe all over the various GND pins across the board, make sure you don't have any shorts.

Refer to the schematic for an idea of how things are connected and work your way through testing each part:
https://github.com/srg74/WLED-wemos-shield/blob/master/resources/Images/schematic.pdf

If you have them, you can use dupont wires to connect the ESP to the shield, then you can jumper GPIO pins to the board and selectively test as you go.  Jumping IO19 (relay) to 3.3v *should* make it click, for example.  If it doesn't, then something might not be right with the 2N7000 or it's connections. Then probe past it and make sure you're seeing 5v to the relay. So on and so forth.  

Good luck!

srg74, 2023-04-08:
> Going to start simple, as this is the most complex assembly I have yet attempted. Ordered PCBs, ordered parts (LCSC, Digikey, and on hand), using AITRIP D1 Mini ESP32, have Tindie 16MB version on [...]

If you want quick resolution of you issue please ring me in WLED discord @serg74#5959

timmgleason, 2023-04-08:
Solved. I had not set my pins properly in WLED for my D1 mini board.
Using this board (https://www.amazon.com/gp/product/B08L79192D) AITRIP D1 Mini, setting are as follows:
LED Pin 16, Relay pin 19, Button pin 17.
[image]
[image]

closed 2023-04-08 by timmgleason

srg74, 2023-04-08:
Thank you for posting your solution.

drewmoseley, 2023-06-03:
FWIW, these GPIO settings worked for me on this device: https://www.amazon.com/dp/B09C5RDZ8G?psc=1&ref=ppx_yo2ov_dt_b_product_details.

Now to figure out the Mic settings.