srg74/WLED-wemos-shield #107: Can't connect to ESP32 while it's on the shield

Kickm3, 2023-04-14:
Hi, and thanks for your work.

I ordered a bunch of v3 boards and parts. I've put two of them together, one with lots of options and one bare minimum.

What works: The pre-check tests are all OK. When I plug the ESP32 with usb (no shield), it turns on and connects to the wifi. When I use the same board on the shields, the power led is on both on the ESP32 and the shield (not the bare minimum of course because it doesn't have one). 

What doesn't work: When plugged in the shield, the ESP32 doesn't connect to wifi (neither the configured network nor in AP mode). The relay makes no sound, the + output for the leds is not powered. I've tried the reset button on the ESP32, doesn't help.

I've tried several 0.14 firmwares (some stock and some compiled myself), different wifi and pin settings... I can't think of anything else to try. Any suggestion?

[image]
[image]
[image]
[image]

srg74, 2023-04-14:
That is sufficient. Please move this discussion to WLED discord. I canceled answer there in more efficient way.

Kickm3, 2023-04-17:
Quick recap of our chat:

I flashing a no-mod firmware after erasing flash memory: no change.
I soldered breadboard cables to TX and RX, and used a ground pin to plug a USB-to-serial reader. This gave me logs at boot:

With ESP not plugged in shield:
`rst:0x1 (POWERON_RESET),boot:0x3b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
�Ada`

With ESP32 plugged in shield:
`rst:0x1 (POWERON_RESET),boot:0x2b (HSPI_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x2b (HSPI_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x2b (HSPI_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0xb (HSPI_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0xb (HSPI_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download`

So when put on shield, the board starts in download mode, waiting for a firmware upload, as if I pushed the (nonexistent) BOOT button. 

On these ESP32, the GPIO0 pin is used as the BOOT button: when LOW it's pushed, when HIGH (3.3V) it's not. Usually it's HIGH by default (using built-in resistors), and for some reason it stayed LOW (<1V) on mine. So I soldered a wire from 3.3V to GPIO0 on the shield and bam, problem solved.

I'll try with another brand of ESP32 to see if this one is problematic, and update here.

Serg, thanks again for your help troubleshooting.

srg74, 2023-04-17:
Better not to do with direct connection to 3V rail but with pull up resistor 4.7k

closed 2023-05-05 by github-actions[bot]

petr-stupka, 2024-05-21:
I found out that if i'll cut the pins like on the [picture](https://github.com/srg74/WLED-wemos-shield/raw/master/resources/Images/wiki/placement2.jpg) so 1x8,1x10 and 2x8, then i do not have such problem.

If i use ESP32 with 2x10 on both sides, i got the same problem.

[image]

Now clue why it is so, i'm just beginner :)

srg74, 2024-05-21:
Thanks. I’ll check if this is related to those pins

spad, 2024-12-04:
I have the same issue with Wemos D1 ESP32. On the same pcb, using d1 mini everithing is fine (of course with the proper pin configuration changes).
The ESP32 works fine when out of the board. When plugged there is no wifi, power led is very low in light. No relay "click". The push button lights the D1 led output (random colors) but pressing it again does not power them off.

srg74, 2024-12-04:
Most likely you have esp board with slightly different pinout than usual.

spad, 2024-12-04:
No, I've checked. It seems the problem is my board cannot be powered by Vcc pin, only by usbc port

srg74, 2024-12-04:
There is a diode between usb port and VCC pin. Short that diode and see if you have power

spad, 2024-12-05:
Thanks srg74. Me too I found this fix on expressif github repo (https://github.com/espressif/arduino-esp32/issues/1972). Unfortunately it looks like my board has some other issue than the diode. As far I can see in my board there is a resistor above the diode not present in the board of that post.
I wonder if there is a store where we can buy the wemos esp32 without this hw bug.

[image]