srg74/WLED-wemos-shield #108: Standard build of the esp32 firmware without any fancy add-on or shield?

DwaineGarden, 2023-05-18:
There use to be a esp32 build with not fancy add-ons or shields.     I can't tell which one is for a simple ESP32. with no options.

https://github.com/srg74/WLED-wemos-shield/tree/master/resources/Firmware/%40Aircoookie/v0.14.0-b2

srg74, 2023-05-19:
No options, just update pins regarding your setup : https://github.com/srg74/WLED-wemos-shield/blob/master/resources/Firmware/%40Aircoookie/v0.14.0-b2/mini_shield/WLED_0.14.0-b2_esp32_mini_shield_minimum_4mb.bin

DwaineGarden, 2023-05-19:
ok thanks.....

Morcegolas, 2023-05-25:
> No options, just update pins regarding your setup : https://github.com/srg74/WLED-wemos-shield/blob/master/resources/Firmware/%40Aircoookie/v0.14.0-b2/mini_shield/WLED_0.14.0-b2_esp32_mini_shield_minimum_4mb.bin

Hello, first of all thanks for sharing this project with others. 

I build one board for now but have material for build more, I'm trying to put this one to work first and then I will build the others for the LEDs I want to use in my house. 

I'm using this cheap board for now but I might buy another now that I saw your recommendations 
[€ 1,39  17％ Off | D1 Mini ESP8266 ESP-12 ESP-12F CH340G V2 USB WeMos D1 Mini WIFI Development Board D1 Mini NodeMCU Lua IOT Board 3.3V With Pins
https://a.aliexpress.com/_mM3XQv2](D1 Mini v4.0.0)

I have many questions, but my first two are:
 - Do I need to update the pins for my board? If yes how do I do that?
 - When I connect the power to the board I ear the relay opening, but when I turn the power off/on in the app/web ui I don't hear anything, is it normal?

Thanks once more for the great project.

srg74, 2023-05-26:
You have assign pins in settings. All pinouts available in wiki.

Morcegolas, 2023-05-26:
Ok it's in the led settings, but I only find 4 gpio settings:

- Led gpio
- button gpio
- IR gpio
- Relay gpio

LEDs are working, button is also working but the relay I already changed to GPIO12 inverted or not it's not making the clicking noise when I turn on/off.

Morcegolas, 2023-05-26:
I already found and changed SCL and SDA GPIOs, they are in the Usermods. But I still haven't figured out why relay isn't working. 😔

srg74, 2023-05-26:
Look for your pins in here https://github.com/srg74/WLED-wemos-shield/wiki choose your board esp8266 or esp32 and follow GPIO numbers.

Morcegolas, 2023-05-26:
I already done that, that's where I found the info, but the relay doesn't click. It has to be esp8266 board because it only goes to GPIO16.

Morcegolas, 2023-05-26:
I used an 'hf3ff 005-1zst' relay but its also 5V 10A 5Pin, and it clicks when I power up the board, I don't think the problem is there.

Morcegolas, 2023-05-26:
The only thing different that I make was flashing the firmware, I didn't found bootloader.bin only bootloader.elf. So I flashed the firmware using ESP-Flasher-dev-macOS using only the firmware file. 
If I need to flash with ESPtool, bootloader, partitions file and firmware could you please 🙏🏼 show me the links of the files that I need to use?
Thanks 🙏🏼

srg74, 2023-05-26:
If you need more help please move to https://discord.gg/Qkt29cG4 my nick @Serg74#5959

closed 2023-06-02 by srg74