srg74/WLED-wemos-shield #111: error with WLED_0.14.0-b3_esp8266_dallas_mod.bin

Francesco-Rugg, 2023-06-21:
I'm trying to flash **0.14.0-b3_esp8266_dallas_mod.bin** on my wemos D1, but every time I got this error

[image]
 
but If I use the bin file **0.13.3_esp8266_display_dallas.bin** everythings works as expected

srg74, 2023-06-21:
Backup you presets and settings, erase your board and upload new bin. 0.14 can’t go on top of 0.13

Francesco-Rugg, 2023-06-22:
Ok, thank you I will try and let you know

Francesco-Rugg, 2023-06-29:
> Backup you presets and settings, erase your board and upload new bin. 0.14 can’t go on top of 0.13

OK, so I got it. 
The problem with 0.14 version is the AP mode that cause this issue.
First things to do with 0.14 version is to connect the wled to a network and the error will not appear!

thanks @srg74

closed 2023-06-29 by Francesco-Rugg