srg74/WLED-wemos-shield #115: Alexa Support in WLED_0.14.0_sp511e_2m.bin

detafun, 2023-11-09:
Hallo,
bitte den Alexa Support mit einbauen. Wäre echt klasse.
Ich habe den sp511e

Danke

Hello,
Please include Alexa support. Would be really great.
I have the sp511e

Thanks

cu Deta

srg74, 2023-11-09:
Hallo,
Bitte laden Sie die neueste Binärdatei herunter

detafun, 2023-11-09:
> Hallo, Bitte laden Sie die neueste Binärdatei herunter

Ich hab alle probiert, immer kein Alexa Support.

detafun, 2023-11-09:
Ha danke gerade gesehen, das ein Update gemacht hast. Super.
Wie sieht es eigentlich aus mir dem Sound Efekten?

srg74, 2023-11-09:
Ich verwende SP511E nicht.

detafun, 2023-11-09:
ok, der hat ein eingebautes Mic, früher gab es mal WLED mit soundReactive. Nur noch in einer alten Version. Und die hat Bugs ALexa zwar da , wird aber nicht gefunden, FB funktioniert nicht richtig....

detafun, 2023-11-09:
Ich komme so langsam dahinter, Aber das Problem ist, das noch was nicht aktiviert ist.
This firmware build does not include simplified UI support

`Server description: 
WLED-SoundReactive

Sync button toggles both send and receive: 
This firmware build does not include simplified UI support.
The following UI customization settings are unique both to the WLED device and this browser.
You will need to set them again if using a different browser, device or WLED IP address.
Refresh the main UI to apply changes.`

srg74, 2023-11-09:
All of the features can be set over settings. Please visit WLED Discord channel for further discussions. I only providing complimentary bin files for various configurations.

closed 2023-11-09 by srg74

detafun, 2023-11-09:
Danke